google/brax #402: Workarounds for Mujoco Menagerie Models

vyeevani, 2023-10-11:
One of the most attractive things to me about brax is the possibility of highly parallelized sim-to-real applications for robotics. However, one of the issues that I'm facing right now that mujoco-menagerie models require a great deal of handholding and editing be importable into Brax due to specific things not being supported in Brax.

The list of things that I've currently faced are:
- elliptical friction
- geometry priorities
- cylinders with depth

It would be great to have a "permissive" mode for mjcf loading in brax.io to handle loading files that may have some surface level violations like the above. (The last one is a bit more tricky but I figure capsule and cylinder aren't terribly far off)

I've got a hacky script to handle the last one:
```
import xml.etree.ElementTree as ET

def cylinder_to_capsule_element(cylinder_elem):
    size = [float(val) for val in cylinder_elem.attrib['size'].split()]
    pos = [float(val) for val in cylinder_elem.attrib.get('pos', "0 0 0").split()]
    quat = cylinder_elem.attrib.get('quat', "1 0 0 0")
    
    from_point = [pos[0], pos[1], pos[2] - size[1]]
    to_point = [pos[0], pos[1], pos[2] + size[1]]

    capsule = ET.Element('geom', 
                         size=str(size[0]), 
                         fromto=f"{from_point[0]} {from_point[1]} {from_point[2]} {to_point[0]} {to_point[1]} {to_point[2]}", 
                         quat=quat)

    return capsule

def modify_xml_file(input_filename, output_filename):
    tree = ET.parse(input_filename)
    root = tree.getroot()

    for geom in root.findall(".//geom[@type='cylinder']"):  # Search for geoms of type cylinder
        capsule = cylinder_to_capsule_element(geom)
        geom.getparent().replace(geom, capsule)  # Replace cylinder with the new capsule element

    tree.write(output_filename)

# Call the function
modify_xml_file('input.xml', 'output.xml')
```

btaba, 2023-10-11:
Hi @vyeevani , we will have an announcement in the coming weeks that makes brax more interoperable with MuJoCo, stay tuned on the announcements page!
We also plan to add "brax-compatible" XMLs to menagerie in the coming weeks/months

JoeMWatson, 2023-10-11:
Are you able to comment on how performant these more complex models will be in brax? 

I have recently tried simulating the Franka Panda manipulator in Brax by taking the Menagerie XML and making it Brax compliant. While the model loaded successfully, jitting the step and reset functions were incredibly slow (on my M1 Macbook) and I got several warnings such as 
```
********************************
[Compiling module jit_reset] Very slow compile? If you want to file a bug, run with envvar XLA_FLAGS=--xla_dump_to=/tmp/foo and attach the results.
********************************
```

This surprised me as the Google Barkour experimental model JITs quite happily, as do the complex Gym-based environments like the Humanoid.  

Do you see there being future performance improvements in this setting, or is it just a consequence of the more complex kinematic chain and collision models?

btaba, 2023-10-15:
Hi @JoeMWatson ,  I'm somewhat surprised Google Barkour JITs happily. Generally, Brax does not scale well with the # of contacts, esp. if they are convex<>convex collisions. I suggest turning off self-collisions for the panda arm (or if you need self-collisions, replacing the meshes with primitives like capsules). Panda may be more expensive if the meshes are larger.

Check out the fork of Google Barkour [here](https://github.com/google/brax/tree/main/brax/experimental/barkour_v0), which only keeps feet<>plane contacts, sufficient for learning joystick policies