google/brax #410: AttributeError while creating Brax Systems

brn-dev, 2023-10-19:
I wanted to try out your colab notebooks (e.g. https://colab.research.google.com/github/google/brax/blob/main/notebooks/basics.ipynb#scrollTo=7GhvlPFTCbuq), but when I'm executing the second cell where Brax Systems are created I get the following error:

[/usr/local/lib/python3.10/dist-packages/brax/io/mjcf.py](https://localhost:8080/#) in loads(xml, asset_path)
    523   mj = mujoco.MjModel.from_xml_string(xml, assets=assets)
    524 
--> 525   return load_model(mj)
    526 
    527 

[/usr/local/lib/python3.10/dist-packages/brax/io/mjcf.py](https://localhost:8080/#) in load_model(mj)
    269   if (mj.geom_priority[0] != mj.geom_priority).any():
    270     raise NotImplementedError('geom_priority parameter not supported.')
--> 271   if mj.opt.collision == 1:
    272     raise NotImplementedError('Predefined collisions not supported.')
    273   q_width = {0: 7, 1: 4, 2: 1, 3: 1}

AttributeError: 'mujoco._structs.MjOption' object has no attribute 'collision'

The collision attribute was removed in this commit of mujoco:
https://github.com/google-deepmind/mujoco/commit/9cf1f6eba42516bee1a051d193d7f511e3bce932

Is there a workaround to fix this error?

brn-dev, 2023-10-19:
The offending line can be found here: https://github.com/google/brax/blob/main/brax/io/mjcf.py#L271

Can it simply be removed now, since the attribute doesn't exist anymore and it would only raise an error anyway?

btaba, 2023-10-19:
Thanks @brn-dev for the fix!

closed 2023-10-19 by btaba