google/brax #420: Adding rangefinder sensors to a robot

eleninisioti, 2023-10-31:
I would like to add rangefinder sensors on a robot. They are available according to the mujoco docs (https://mujoco.readthedocs.io/en/stable/XMLreference.html#sensor) but I cannot find any examples out there 

I tried to do it by modifying the ant xml with a tag ` <site name="range_finder_1" pos="0.25 0 0.5"/>` inside the robot body and a tag `<sensor name="range_finder_1_sensor" site="torso:range_finder_1" type="range"/>` but am getting an error that sensor is undefined when calling `mujoco.MjModel.from_xml_string`

btaba, 2023-10-31:
Hi @eleninisioti , Brax doesn't support sensors, you should post here for help creating the rangefinder sensor:  https://github.com/google-deepmind/mujoco

closed 2023-10-31 by btaba

varunagrawal, 2025-04-07:
Since Mujoco 3.2.3, sensor support has been added to MJX, so if you upgrade Mujoco, you should now be able to access sensor information via `pipeline_state.sensordata`.

However, this is a single array, so you need the indices to extract the necessary information. If you know the sensor ID number `sensor_idx`, the following function will get you what you need:

```python
def get_sensor_value(model, data, sensor_idx):
    name = mjx.id2name(model, mujoco.mjtObj.mjOBJ_SENSOR, sensor_idx)
    start = model.sensor_adr[sensor_idx]
    dim = model.sensor_dim[sensor_idx]
    value = data.sensordata[start:start + dim]
    return name, value
```

Here `model` is `env.sys` (aka the `sys` variable in your environment) and `data` is the `pipeline_state`.