google/brax #421: Add Gymnasium Wrapper

vyeevani, 2023-11-10:
Lots of other frameworks seem to be switching to gymnasium. It'd be nice to also have a gymnasium wrapper in brax.

btaba, 2023-11-10:
Hi @vyeevani, is this what you're looking for or something else? 

https://github.com/google/brax/blob/16304037a36b1d9c8c0b3084f57d1159627b636b/brax/envs/wrappers/gym.py#L27

vyeevani, 2023-11-10:
It’s that except using Gymnasium instead of Gym. The imports would change and also the returns are different for the step API. For example, you’d need to also return truncations (that you can get from the EpisodeWrapper).

https://github.com/google/brax/blob/16304037a36b1d9c8c0b3084f57d1159627b636b/brax/envs/wrappers/gym.py#L27

vyeevani, 2023-11-10:
https://gymnasium.farama.org

vyeevani, 2023-11-10:
It's a simple enough lift. It seems like OpenAI Gym isn't going to be maintained anymore with the aim of switching everyone over to Gymnasium.

btaba, 2023-11-10:
Ok, yeah the APIs are subtly different. Feel free to open up a PR with `brax/envs/wrappers/gymnasium.py`