google/brax #422: Increase modularization of training wrappers

vyeevani, 2023-11-12:
The training wrappers for auto reset and episode wrapper are leaking info to each other.

This is a bigger problem if people want to stack their own wrappers. For example, I'd like to write a meta-episode wrapper that takes multiple episodes and aggregates them into a single meta episode for use in a meta-RL setting. The wrapper that I'm writing needs to track the number of episodes so that it can do a meta episode reset when it reaches some watermark. However, the auto reset doesn't seem to work for this.

My thinking is that caching the initial state.info is enough for

vyeevani, 2023-11-12:
Haven't tested this, but I'm thinking something like this:

`class AutoResetWrapper2(Wrapper):
  """Automatically resets Brax envs that are done."""

  def reset(self, rng: jax.Array) -> State:
    base_state = self.env.reset(rng)
    info = base_state.info.copy()
    info.update({
        'initial_base_state': base_state,
        'current_base_state': base_state
    })
    
    return State(
        pipeline_state=base_state.pipeline_state,
        obs=base_state.obs,
        reward=base_state.reward,
        done=base_state.done,
        metrics=base_state.metrics,
        info=info
    )

  def step(self, state: State, action: jax.Array) -> State:
    initial_base_state = state.info['initial_base_state']
    initial_info = initial_base_state.info.copy()
    initial_info.update({
        'initial_base_state': initial_base_state,
        'current_base_state': initial_base_state
    })
    initial_state = State(
        pipeline_state=initial_base_state.pipeline_state,
        obs=initial_base_state.obs,
        reward=initial_base_state.reward,
        done=initial_base_state.done,
        metrics=initial_base_state.metrics,
        info=initial_info
    )

    current_base_state = state.info['current_base_state']
    next_base_state = self.env.step(state, action)
    next_info = next_base_state.info.copy()
    next_info.update({
        'initial_base_state': initial_base_state,
        'current_base_state': next_base_state
    })
    next_state = State(
        pipeline_state=next_base_state.pipeline_state,
        obs=next_base_state.obs,
        reward=next_base_state.reward,
        done=next_base_state.done,
        info=next_info,
    )

    def where_done(x, y):
      return jp.where(done, x, y)

    return jax.tree_map(
        where_done, initial_state, next_state
    )`

vyeevani, 2023-11-13:
This allows us to pass back information from the wrapped things up to the clients that are expecting them (episode wrappers truncated for instance) while at the same time caching this stuff so that we can reuse it during the reset.

vyeevani, 2023-11-13:
Tested the above (not very rigorously), seems to work

btaba, 2023-12-08:
Hi @vyeevani 

Thanks for the proposal, indeed there is logic leaking into reset from the episode wrapper.

If you changed `first_pipeline_state` to `first_state` (which would contain the first `State` object) in the impl at HEAD, would that suffice for your use-case? Why do you need to store `current_base_state` in the info as well?

My only concern beyond cleaner semantics is how this affects performance