google/brax #427: Mujoco environment support

reginald-mclean, 2023-11-22:
Hi!
I am the Project Manager of [Meta-World](https://github.com/Farama-Foundation/Metaworld/) and we are looking to add support for Brax (while we wait for Mujoco-MJX - there's some collisions and other things not fully supported yet). However, we use a weld equality constraint for a mocap on the end-effector. This allows us to use an action space of size 4, where 1-3 are the desired xyz coordinates of the mocap. I am wondering if you have any suggestions for methods to modify the environments to remove the mocap but still use a similar action space. Thanks!

btaba, 2023-12-08:
Hi @reginald-mclean , what are the blockers on using MuJoCo-MJX out of curiosity? 

What environment are you describing with this 4-dof action space? Can you share an XML?

btaba, 2024-02-11:
Closing due to inactivity

closed 2024-02-11 by btaba