google/brax #435: Apply external force to the agent

aliang8, 2023-12-13:
Hi,

I was wondering if anyone has pointers on how to apply an external force to an agent, specifically the HalfCheetah environment. 

I read in the MuJoCo documents there is a `xfrc_applied` field that implements this. Is there a similar feature for BRAX environments? Thanks!

aliang8, 2023-12-19:
Bumping this!

btaba, 2023-12-22:
Hey @aliang8 , there isn't a similar feature in brax right now, but have you checked out [MJX](https://mujoco.readthedocs.io/en/stable/mjx.html)? We aim to have feature parity with MuJoCo using that physics engine (which is now also ported into brax via https://github.com/google/brax/commit/f0cf1092201f66724bf4d38f3b222beaa5e208f3)

closed 2023-12-22 by btaba