google/brax #437: Discrepancy in Observation Dimension for Humanoid Environment

Zhenyu2Liang, 2023-12-26:
Hi,

I've encountered a couple of issues while working with the **humanoid** environment in Brax that I'd like to discuss.

1. **Details on 244-Dimensional Observation**: In my use of the humanoid environment, I've noticed that the observation vector is 244 dimensions, whereas the documentation or related papers mention a 299-dimensional observation.  Could you provide an explanation for this discrepancy?  Additionally, I would appreciate if you could detail what components or features are included in the 244-dimensional observation vector.

```python
    from brax.envs import get_environment
    import jax

    env = get_environment(env_name="humanoid")

    state = env.reset(jax.random.PRNGKey(42))
    action = jax.random.uniform(jax.random.PRNGKey(43), shape=(17, )) * 0.8 - 0.4
    state = env.step(state, action)

    print(state.obs.shape)
 ```

   Upon executing this, I found the shape of `state.obs` to be 244 dimensions. 

2. **Request for Detailed Documentation or Guide**: I am interested in developing with Brax but have found that, unlike JAX, it lacks detailed documentation or tutorials. This is particularly challenging for beginners, as many aspects of Brax are not immediately clear or intuitive. Is there a plan to create more comprehensive documentation or a guide to help new users understand and effectively use Brax?

Thank you for your assistance and for the great work on this project!

btaba, 2024-01-02:
Hi @Zhenyu2Liang 

The documentation on the humanoid observation space can be found here:

https://github.com/google/brax/blob/222eeb94a8582c1171ec3382cf782e0ed6cc3f76/brax/envs/humanoid.py#L71-L140

or you can check the code pointer here

https://github.com/google/brax/blob/222eeb94a8582c1171ec3382cf782e0ed6cc3f76/brax/envs/humanoid.py#L321-L328

The main difference with the humanoid version in gymnasium or other repositories is that we don't include external contact forces, as noted in the previous code pointer. See 

https://github.com/Farama-Foundation/Gymnasium/blob/046c76f623675e3bf4c43e701e025c676d0b420f/gymnasium/envs/mujoco/humanoid_v5.py#L476-L485

RE documentation, noted that the documentation is a bit lacking, what kind of tutorials would you like to see that are missing?

closed 2024-01-02 by btaba

Zhenyu2Liang, 2024-01-03:
Thank you very much for your prompt response. I greatly appreciate it.