google/brax #451: MjxEnv no where to be found

Engineering-Geek, 2024-02-08:
[image]
As shown in the screen shot, MjxEnv is no longer detected anywhere in the code. I swear it was working earlier today on my PC...
When I open the file and looked through the code and current main branch, I can't find MjxEnv anywhere now unless if I am blind.

btaba, 2024-02-08:
Hi @Engineering-Geek we just did a release and removed MjxEnv. You shouldn't need it anymore! If you'd like to work off of an older version of the code base, you can still use `0.9.4` via `pip install`

btaba, 2024-02-08:
For more background info, you can use `PipelineEnv` now and add `backend='mjx'`, and use the MJX backend like any other backend in Brax

closed 2024-02-08 by btaba