google/brax #469: HTML rendering not working in local Jupyter Notebook

hoangsep, 2024-03-16:
Hi, I am trying to run the [tutorial training notebook](https://colab.research.google.com/github/google/brax/blob/main/notebooks/training.ipynb) locally. However the HTML rendering seems to be quite flickering. The first render stacked on top of each other 
[simplescreenrecorder-2024-03-16_11.38.05.webm](https://github.com/google/brax/assets/3454586/156e5221-d388-436d-9b33-8a637b0e07d7)
while the second render just outputing nothing

[image]

btaba, 2024-03-23:
Hi @hoangsep , I'm not able to reproduce the issue with the tutorial notebook on colab.research.google.com

Can you try running with chrome developer tools, and inspecting any javascript errors https://developer.chrome.com/docs/devtools/open ?

hoangsep, 2024-03-23:
Hi @btaba , I meant I got the issue when I run it locally, on my laptop, and not on google colab.
Apologies for any confusion.

btaba, 2024-10-04:
Closing issue due to lack of reproducibility. Feel free to open another issue