google/brax #470: Failure to write videos upon training completion

HowdyMoto, 2024-03-18:
I'm on MacOS Sonoma, and when training completes, I get:
```
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/os.py", line 225, in makedirs
    mkdir(name, mode)
OSError: [Errno 30] Read-only file system: '/saved_videos'
```
I've done `chmod -R u+w .` from my brax folder, 
I've set up brax using `pip install -e .` from my local repo,
and finally, I'm running `sudo learn humanoid`, so I thought that would give it permissions to write the output video wherever it wishes.

Any tips on how to stop it from failing\? It's unclear to me where it's trying to write to by default.

btaba, 2024-03-23:
Hi @HowdyMoto , it looks like it's try to write to '/saved_videos', so the root directory

Can you specify a `logdir` in your command? See https://github.com/google/brax/blob/2329ae76759e37b0b1f1861cf34e5a67d0f7efa8/brax/training/learner.py#L74C22-L74C28

Feel free to send over a PR to set a default `logdir`, perhaps `/tmp/brax`, possibly creating the directory if empty.

btaba, 2024-10-04:
Closing issue due to inactivity

closed 2024-10-04 by btaba