google/brax #480: Optimizer with MultiTransform throws ValueError

varunagrawal, 2024-04-19:
I want to freeze parts of my network for training, so to do this, I modified the `ppo.train` function to accept an optimizer object.

I then define the optimizer as
```python
def zero_grads():
      def init_fn(_):
          return ()
      def update_fn(updates, state, params=None):
          return jax.tree_map(jp.zeros_like, updates), ()
      return optax.GradientTransformation(init_fn, update_fn)

op = optax.multi_transform(
    {
        "adam": optax.adam(3.0e-4),
        "zero": zero_grads()
    }, {
        "Encoder_0": "zero",
        "BCModel_0": "zero",
        "ResidualPolicy_0": "adam"
    })
```

However, when the optimizer calls init as `optimizer.init(init_params)` when defining the TrainingState, it throws the following error:
```
ValueError: Expected dict, got PPONetworkParams(policy={'params': ...
```

This does not happen when I define the optimizer as `op = optax.adam(3.0e-4)`. Since both variants are `optax.GradientTransformationExtraArgs`, can someone please explain what is happening and how can I resolve this?

Here's the full stack trace:
```
Traceback (most recent call last):
  File "/Users/scripts/train.py", line 181, in <module>
    main()
  File "/Users/scripts/train.py", line 79, in main
    tx = optimizer.init(init_params)
  File "/Users/.pyenv/versions/3.10.12/lib/python3.10/site-packages/optax/_src/combine.py", line 214, in init_fn
    inner_states = {
  File "/Users/.pyenv/versions/3.10.12/lib/python3.10/site-packages/optax/_src/combine.py", line 217, in <dictcomp>
    mask_compatible_extra_args=mask_compatible_extra_args).init(params)
  File "/Users/.pyenv/versions/3.10.12/lib/python3.10/site-packages/optax/_src/wrappers.py", line 545, in init_fn
    masked_params = mask_pytree(params, mask_tree)
  File "/Users/.pyenv/versions/3.10.12/lib/python3.10/site-packages/optax/_src/wrappers.py", line 509, in mask_pytree
    return jax.tree_util.tree_map(
  File "/Users/.pyenv/versions/3.10.12/lib/python3.10/site-packages/jax/_src/tree_util.py", line 311, in tree_map
    all_leaves = [leaves] + [treedef.flatten_up_to(r) for r in rest]
  File "/Users/.pyenv/versions/3.10.12/lib/python3.10/site-packages/jax/_src/tree_util.py", line 311, in <listcomp>
    all_leaves = [leaves] + [treedef.flatten_up_to(r) for r in rest]
ValueError: Expected dict, got PPONetworkParams(policy={'params': {'ResidualPolicy_0': {'Dense_0': {'kernel': Array([[-0.081,  0.018, -0.062, ...,  0.081,  0.015,  0.01 ],
       [ 0.018,  0.075,  0.039, ...,  0.035,  0.072, -0.018],
       [-0.046,  0.078,  0.027, ...,  0.038,  0.04 , -0.08 ],
       ...,
       [ 0.012,  0.029, -0.009, ...,  0.01 , -0.017,  0.022],
       [ 0.007,  0.034,  0.002, ..., -0.087,  0.059,  0.08 ],
       [ 0.084,  0.024, -0.017, ..., -0.061, -0.087,  0.007]], dtype=float32), 'bias': Array([0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.,
       0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.,
       0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.,
       0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.,
       0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.,
       0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.], dtype=float32)}, 'Dense_1': {'kernel': Array([[ 0.079,  0.092,  0.13 , ...,  0.01 , -0.024, -0.124],
       [ 0.037, -0.038,  0.072, ...,  0.13 ,  0.051, -0.032],
       [-0.087, -0.007, -0.026, ...,  0.051,  0.071, -0.025],
       ...,
       [ 0.033, -0.073, -0.038, ...,  0.154,  0.025,  0.062],
       [-0.182, -0.127,  0.011, ..., -0.003, -0.08 ,  0.029],
       [-0.115, -0.125,  0.094, ..., -0.041, -0.002, -0.139]], dtype=float32), 'bias': Array([0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.,
       0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.,
       0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.,
       0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.,
       0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.,
[... 324 lines cut ...]
       [ 0.015],
       [ 0.064],
       [-0.006],
       [-0.008],
       [ 0.068],
       [-0.014],
       [ 0.065],
       [ 0.034],
       [ 0.002],
       [ 0.014],
       [-0.103],
       [ 0.082],
       [-0.088],
       [-0.029],
       [-0.006],
       [ 0.083],
       [ 0.063],
       [-0.049],
       [ 0.099],
       [-0.012],
       [-0.018],
       [ 0.077],
       [ 0.054],
	   [-0.095]], dtype=float32), 'bias': Array([0.], dtype=float32)}}}).
```

varunagrawal, 2024-04-19:
NOTE: I got the idea for using the `optax.multi_transform` from https://github.com/google/flax/discussions/1706.

btaba, 2024-04-23:
Hi @varunagrawal , I'm not familiar with `optax.multi_transform`, but `optimizer.init` is expecting a dict, and `PPONetworkParams` is not a dict. Maybe try `optimizer.init(init_params.__dict__)`?  Will move this to a discussion for now