google/brax #483: Documentation bug

Velythyl, 2024-05-02:
I'm currently doing a project related to transferring control policies from simulator to simulator. This lead me to try to standardize action and observation spaces between brax and mujoco.

I implemented wrappers that map the obs and action spaces from mujoco to brax, according to each's table description

- brax: https://github.com/google/brax/blob/main/brax/envs/ant.py
- mujoco: https://gymnasium.farama.org/environments/mujoco/ant/

I found out that gymnasium's mujoco and gym's mujoco report different index orders for the action and observation spaces.
https://github.com/Farama-Foundation/Gymnasium/issues/1035

I then tried completely removing my mapping wrapper (maps mujoco's obs and action orders to brax's), only to find that my agent transfers a LOT better to mujoco from brax.

All this to say, I'm fairly sure that brax's doc reports the wrong ordering for the action and observation spaces (at least for Ant) and that gymnasium's doc should by copy-pasted into brax

btaba, 2024-05-03:
Thanks for the bug report! Please feel free to open up a PR!

Velythyl, 2024-05-24:
Hey! I'll get on that a bit later, been very busy with my own stuff lately

Another report: Humanoid reports a 376-space obs in both gymnasium and brax. For gymnasium it is true, for brax it is actually 244-spaced

[image]

[image]

Velythyl, 2024-05-24:
Same thing for HumanoidStandup