google/brax #485: `mjx.ncon` removed as of MuJoCo 3.1.5

namheegordonkim, 2024-05-13:
Hi friends,

It appears that MuJoCo officially removed `mjx.ncon` (https://mujoco.readthedocs.io/en/latest/changelog.html#version-3-1-5-may-7-2024) and hence this line is broken:

https://github.com/google/brax/blob/b45760cfa6ba0c3567ae36f988766a8299853c49/brax/contact.py#L38

Should be a quick fix; perhaps L38 should be replaced by

```
mujoco.MjData(sys.mj_model).ncon
```

or something cleverer.

btaba, 2024-05-13:
Hi @namheegordonkim , we have a release in the backlog that we'll push out shortly! Apologies

btaba, 2024-05-13:
See https://github.com/google/brax/releases/tag/v0.10.4 for the latest release. Thanks for the ping @namheegordonkim

closed 2024-05-13 by btaba