google/brax #519: Trained model inference

OmurAydogmus, 2024-09-18:
I need to use the trained network in a separate test code without using the training-side code. I have a problem with the shape of **inference_fn** as below:

```
# training code
inference_fn
Out[10]: <function brax.training.agents.ppo.networks.make_inference_fn.<locals>.make_policy.<locals>.policy(observations: jax.Array, key_sample: jax.Array) -> Tuple[jax.Array, Mapping[str, Any]]>
```
```
# test code
inference_fn
Out[181]: <function brax.training.agents.ppo.networks.make_inference_fn.<locals>.make_policy(params: Tuple[Any, Any], deterministic: bool = False) -> brax.training.types.Policy>
```
Test code:
 ```
# Load the saved parameters
model_path = '/tmp/mjx_brax_quadruped_policy'
params = model.load_params(model_path)

# Create the environment (ensure this matches the original environment from Code1)
observation_size = env.observation_size
action_size = env.action_size

# Re-create the policy and value networks with the exact same architecture
make_networks_factory = functools.partial(
    ppo_networks.make_ppo_networks,
    observation_size=observation_size,
    action_size=action_size,
    policy_hidden_layer_sizes=(128, 128, 128, 128)  # Match the architecture from Code1
)

# Initialize the PPO networks (policy and value networks)
ppo_modified = make_networks_factory()

# Now use `make_inference_fn` to create the inference function
inference_fn = ppo_networks.make_inference_fn(ppo_modified)

# JIT-compile the inference function
jit_inference_fn = jax.jit(inference_fn(params)) 
```

btaba, 2024-09-18:
Hi @OmurAydogmus, what's the issue exactly? Does the "test" code not work?

OmurAydogmus, 2024-09-18:
Yes, the test code works, but the model does not perform the same way as it does during training. This is because the shapes of the **inference_fn** variables are not the same as shown below:
 
**inference_fn in Training Code**
Out[10]: <function brax.training.agents.ppo.networks.make_inference_fn.<locals>.make_policy.<locals>.policy(observations: jax.Array, key_sample: jax.Array) -> Tuple[jax.Array, Mapping[str, Any]]>

**inference_fn in Testing  Code**
Out[181]: <function brax.training.agents.ppo.networks.make_inference_fn.<locals>.make_policy(params: Tuple[Any, Any], deterministic: bool = False) -> brax.training.types.Policy>

OmurAydogmus, 2024-09-19:
Yes, the test code works. But the model does not give same action as training code side. The full code is belows:

```
import functools
import jax
import os

from datetime import datetime
from jax import numpy as jp
import matplotlib.pyplot as plt

from IPython.display import HTML, clear_output

import brax
 
import flax
from brax import envs
from brax.io import model
from brax.io import json
from brax.io import html
from brax.training.agents.ppo import train as ppo
from brax.training.agents.sac import train as sac

import mediapy as media
 
env_name = 'ant'  # @param ['ant', 'halfcheetah', 'hopper', 'humanoid', 'humanoidstandup', 'inverted_pendulum', 'inverted_double_pendulum', 'pusher', 'reacher', 'walker2d']
backend = 'positional'  # @param ['generalized', 'positional', 'spring']
 
train_fn =  functools.partial(ppo.train,  num_timesteps=50_000_000, num_evals=10, reward_scaling=10, episode_length=1000, 
                              normalize_observations=True, action_repeat=1, unroll_length=5, num_minibatches=32, 
                              num_updates_per_batch=4, discounting=0.97, learning_rate=3e-4, entropy_cost=1e-2, 
                              num_envs=4096, batch_size=2048, seed=1)
 
max_y = 8000
min_y = 0

env = envs.get_environment(env_name=env_name, backend=backend)
state = jax.jit(env.reset)(rng=jax.random.PRNGKey(seed=0)) 

xdata, ydata = [], []
times = [datetime.now()]

def progress(num_steps, metrics):
  times.append(datetime.now())
  xdata.append(num_steps)
  ydata.append(metrics['eval/episode_reward'])
  clear_output(wait=True)
  plt.xlim([0, train_fn.keywords['num_timesteps']])
  plt.ylim([min_y, max_y])
  plt.xlabel('# environment steps')
  plt.ylabel('reward per episode')
  plt.plot(xdata, ydata)
  plt.show()

make_inference_fn, params, _ = train_fn(environment=env, progress_fn=progress)

print(f'time to jit: {times[1] - times[0]}')
print(f'time to train: {times[-1] - times[1]}')
 
model.save_params('/tmp/params', params)
params = model.load_params('/tmp/params')
inference_fn = make_inference_fn(params)

# =============================================================================
# Training Policies 
# =============================================================================
env = envs.create(env_name=env_name, backend=backend)

jit_env_reset = jax.jit(env.reset)
jit_env_step = jax.jit(env.step)
jit_inference_fn = jax.jit(inference_fn)

rollout = []
rng = jax.random.PRNGKey(seed=1)
state = jit_env_reset(rng=rng)
for _ in range(100):
  rollout.append(state.pipeline_state)
  act_rng, rng = jax.random.split(rng)
  act, _ = jit_inference_fn(state.obs, act_rng)
  state = jit_env_step(state, act)

n_steps = 500
render_every = 2

media.show_video(
    env.render(rollout[::render_every], camera='track'),
    fps=1.0 / env.dt / render_every)
# =============================================================================
# Load Policies and Test
# =============================================================================
paramsTEST = model.load_params('/tmp/params')

ppoTEST = ppo.ppo_networks.make_ppo_networks(action_size=env.action_size, observation_size=env.observation_size)
make_inference = ppo.ppo_networks.make_inference_fn(ppoTEST)
inference_fnTEST = make_inference(paramsTEST)
 
env = envs.create(env_name=env_name, backend=backend)

jit_env_reset = jax.jit(env.reset)
jit_env_step = jax.jit(env.step)
jit_inference_fn = jax.jit(inference_fnTEST)

rollout = []
rng = jax.random.PRNGKey(seed=1)
state = jit_env_reset(rng=rng)
for _ in range(100):
  rollout.append(state.pipeline_state)
  act_rng, rng = jax.random.split(rng)
  act, _ = jit_inference_fn(state.obs, act_rng)
  state = jit_env_step(state, act)

media.show_video(
    env.render(rollout[::render_every], camera='track'),
    fps=1.0 / env.dt / render_every)
```

OmurAydogmus, 2024-09-19:
hank you, btaba. I think I found the mistake. Observations need to be normalized. I will try that.

OmurAydogmus, 2024-09-19:
I think it is okay. We nee to define **preprocess_observations_fn** using normalization.

ppoTEST = ppo.ppo_networks.make_ppo_networks(action_size=env.action_size, observation_size=env.observation_size, preprocess_observations_fn=running_statistics.normalize)

closed 2024-09-21 by OmurAydogmus