google/brax #520: training taking to long on local GPU

GandharvMahajan, 2024-09-28:
I am training a quadruped to walk similar to deepmind mujoco tutorials. I am running the training on an Nvidia L4 GPU
Colab Script - [https://colab.research.google.com/drive/1uL1TKBYXOYpeajOiaF1whhsB0SSKPEe3?usp=sharing](url)
the following cell takes a very long time and never finishes execution. I reduced the timesteps to just 10

```
train_fn = functools.partial(
      ppo.train, num_timesteps=10, num_evals=10,
      reward_scaling=1, episode_length=1000, normalize_observations=True,
      action_repeat=1, unroll_length=5, num_minibatches=32,
      num_updates_per_batch=4, discounting=0.97, learning_rate=3.0e-4,
      entropy_cost=1e-2, num_envs=64, batch_size=64,
      network_factory=make_networks_factory,
      randomization_fn=domain_randomize,
      policy_params_fn=policy_params_fn,
      seed=0)

env = envs.get_environment(env_name)
eval_env = envs.get_environment(env_name)
make_inference_fn, params, _= train_fn(environment=env,
                                progress_fn=progress,
                                eval_env=eval_env)
```

on reducing the timesteps, i thought the training would complete faster. I made sure that jax is utilizing GPU. in addition my GPU shows 100% usage in `nvidia-smi`
Any help would be appreciated!

GandharvMahajan, 2024-09-29:
I was using a non mjx compatible robot xml which threw no errors but slowed everything down.

closed 2024-09-29 by GandharvMahajan

uvikas, 2024-11-09:
@GandharvMahajan How did you convert the robot xml to be compatible with MJX? I am also noticing slow down with a custom environment.

btaba, 2024-12-04:
@uvikas you can see some examples [here](https://github.com/google-deepmind/mujoco_menagerie/tree/main/google_barkour_vb#mjx) and [here](https://github.com/google-deepmind/mujoco_menagerie/tree/main/google_barkour_vb#mjx) of models that were modified to work well with MJX/brax