google/brax #538: How to apply APG algorithm in a Brax pipeline environment for SMPLH-based MJCF?

dohunn, 2024-10-20:
Hello there, thanks for amazing work on Brax!

I'm a PhD student in computer vision and machine learning. My current research is motion generation using [SMPLH](https://smpl-x.is.tue.mpg.de/) (SMPL + MANO) based humanoid robots. This text is going to be a bit long, so I'm sorry in advance.

I want to use [APG](https://github.com/google/brax/blob/main/brax/training/agents/apg/train.py) to train a policy that mimics motion capture data like AMASS (SMPLH-based motion dataset) well. Of course, there are other algorithms for motion generation, but I wanted to train a policy quickly using APG without GAN-based rewards or PPOs.

First, I'd like to talk about my background and the problem I'm facing. Before using Brax, I used Isaac Gym as a physics simulation and built an environment based on SMPLH-based robots based on this physics engine. I applied the [AMP](https://arxiv.org/abs/2104.02180) algorithm, which uses GAN-based rewards for SFU data from [AMASS](https://amass.is.tue.mpg.de/), to this environment and successfully learned policies that mimic SFU motion dataset. However, when I applied the MJCF file used by Isaac Gym to Brax based on the Positional backend and ran the APG algorithm, it failed to mimic the motion data.

I tried a few things but couldn't fix it. Here are the issues I observed using Brax on SMPLH-based robot. The APG algorithm visualization was different from the original data. The robot appears to have no movement at all.

- [Original data visualization] 
[image]

- [APG policy visualization] 
[image]

Second, here's what I tried to do to resolve the issue. The MJCF file that used in Isaac Gym caused NaN values when training APG, but I noticed that the NaN values went away after trying 1st thing of the following things.

1. Add “\<custom> ~ \</custom>” provided by https://github.com/google/brax/blob/main/brax/envs/assets/humanoid.xml into MJCF.
2. Adjust joint’s stiffness, damping, and armature.
3. Set gear value of actuator to 1.
4. Adjust sys.actuactor.ctrlrange and sys.actuator.forcerange

I used an MJCF file for the SMPLH robot built with [SMPLSim](https://github.com/ZhengyiLuo/SMPLSim), which works for Isaac Gym and Mujoco. sys.actuator.ctrlrange, sys.actuator.forcerange, and position and velocity gains for PD control were based on the parameters used in SMPLSim. I'm not an expert in robotics and I may have adjusted them incorrectly. But, when I adjusted these parameters, as well as the stiffness, damping, and the armature, I got the same phenomena. Also, I tried MJX as a backend, but it was slow to train APG model and didn't help me as much as a positional backend. I had no issues running the [APG tutorial](https://colab.research.google.com/github/google-deepmind/mujoco/blob/main/mjx/training_apg.ipynb) using [robot dog MJCF](https://github.com/google-deepmind/mujoco_menagerie/blob/main/anybotics_anymal_c/anymal_c_mjx.xml), so I think there's something wrong with the MJCF settings or the environment.

**Any advice or hints on how to fix this? Also, are there any references or comments on what the values set inside <custom> in https://github.com/google/brax/blob/main/brax/envs/assets/humanoid.xml mean and affect the simulation?**

If the MJCF file I used is causing the problem, I thought I could try with a config file suitable for Brax.v1. **However, when I run the [Mujoco converter](https://github.com/google/brax/blob/main/brax/v1/tools/mujoco_converter.py) provided by Brax.v1.tools to convert the MJCF file to a config file for Brax.v1, I get the following error. How can I solve this problem?**

[image]

I also attached the implementation I used as the below. Note that this code has dependencies on Jax CUDA and the Pytorch CPU.

> [smplh_humanoid_apg.zip](https://github.com/user-attachments/files/17449863/smplh_humanoid_apg.zip)

The files in the compressed file have the following meaning.
- Motion file: 0005_Jogging001_poses_smplh.npy
- MJCF file for SMPLH robot: smplh_humanoid_smplsim_brax.xml
- Environment for SMPLH robot: smplh_humanoid_env.py
- Convert MJCF to config: convert_mjcf_to_config.py

I've bolded the questions I'm curious about. Thank you for reading this long text.

Andrew-Luo1, 2024-12-06:
Hi! I think there's two things that make your approach difficult. First, your model has many degrees of freedom, which is probably causing MJX to be slow. Second, you're using the old positional backend. As far as I know, the current APG algorithm has not been tried and tested in either of these two settings.

If you could work with a simplified humanoid, like [the classic](https://github.com/google/brax/blob/main/brax/envs/assets/humanoid.xml) but perhaps with position-controlled joints, and could use the MJX backend, motion mimicking with APG would probably be straight-forward as long as you:
1) Add this flag to the xml:
```
  <option iterations="1" ls_iterations="4" solver="Newton" cone="pyramidal" impratio="100">
    <flag eulerdamp="disable"/>
  </option>
```
2) Use double precision:
```
from jax import config
config.update("jax_enable_x64", True)
config.update('jax_default_matmul_precision', jax.lax.Precision.HIGH)
```

Good luck!

btaba, 2025-01-31:
Closing the issue due to lack of activity.

closed 2025-01-31 by btaba