google/brax #550: Saving intermediate policies during the training

mazzamani, 2024-11-05:
My goal is to save intermediate policies. The policy at the end of training works fine for me:

```python
        # ... main training code
        self.make_inference_fn, self.params, _ = self.train_fn(environment=env, progress_fn=self.progress_callback,
                                                               policy_params_fn=self.policy_params_callback)
        self.visualize_trajectory()
```

However, when I call a `self.visualize_trajectory()` from `self.policy_params_callback` 
```python
        def policy_params_callback(self, step, make_policy, params):
            self.make_inference_fn = make_policy
            self.params = params  
            self.visualize_trajectory()
 ```
I run into the following error:
```python
  File "/brax/training_code.py", line 29, in policy_params_callback
    self.visualize_trajectory()
  File "/brax/training_code.py", line 74, in visualize_trajectory
    act, _ = jit_inference_fn(state.obs, act_rng)
  File "/miniconda3/envs/brax/lib/python3.10/site-packages/brax/training/agents/ppo/networks.py", line 44, in policy
    logits = policy_network.apply(*params, observations)
  File "/miniconda3/envs/brax/lib/python3.10/site-packages/brax/training/networks.py", line 104, in apply
    return policy_module.apply(policy_params, obs)
TypeError: argument of type 'PPONetworkParams' is not iterable
 ```
This is my visualization method which is adapted from the example [training code](https://colab.research.google.com/github/google/brax/blob/main/notebooks/training.ipynb):

```python
    def visualize_trajectory(self):
        inference_fn = self.make_inference_fn(self.params)
        env = self.load_environment()
        jit_env_reset = jax.jit(env.reset)
        jit_env_step = jax.jit(env.step)
        jit_inference_fn = jax.jit(inference_fn)

        trajectory = []
        rng = jax.random.PRNGKey(seed=1)
        state = jit_env_reset(rng=rng)

        for _ in tqdm(range(1000)):
            trajectory.append(state.pipeline_state)
            act_rng, rng = jax.random.split(rng)
            act, _ = jit_inference_fn(state.obs, act_rng)
            state = jit_env_step(state, act)

        rendered_html = html.render(env.sys.tree_replace({'opt.timestep': env.dt}), trajectory)
        with open("trajectory_visualization.html", "w") as file:
            file.write(rendered_html)

        print("trajectory visualization prepared.")
```

Any idea why it is not working?

btaba, 2024-11-16:
Hi, `policy_module.apply` expects the params to be an iterable, not `PPONetworkParams`. You can unpack `params ` via `(params.policy, params.value)`

closed 2024-11-16 by btaba

mazzamani, 2024-11-21:
Thanks for the reply. This works:
```python
inference_fn = self.make_inference_fn((self.params[0], self.params[1].policy))
```