google/brax #571: Providing randomization_fn in ppo train triggers trace error when eval_env not provided

Nate711, 2025-01-13:
Providing `randomization_fn` argument to `brax.training.agents.ppo.train` triggers trace error when `eval_env` argument is not provided.

Probably the offending line has something to do with setting the eval_env to environment and then randomizing multiple times??
https://github.com/google/brax/blob/69637a359463738140c1b850f61ad0088a23538b/brax/training/agents/ppo/train.py#L561

Working example:
https://colab.research.google.com/drive/1VwuZfD3Tc_ndzWSwlTfDLzNGi1CmyYF8#scrollTo=mu9waFpPevPl

Specifically the code
```
# FAILS
make_inference_fn, params, _= train_fn(environment=env,
                                        progress_fn=progress)

# WORKS
# make_inference_fn, params, _= train_fn(environment=env,
#                                         progress_fn=progress,
#                                         eval_env=eval_env,)
```

Nate711, 2025-01-13:
Nevermind! The underlying cause was providing a `policy_params_fn` function that had access to the `environment` used by `ppo.train`. When I provided `policy_params_fn` access to the `eval_env` used for `ppo.train` everything worked.

closed 2025-01-13 by Nate711

avirupdas55, 2025-03-25:
I am facing a similar issue. Let's say you want to update the mass of the pole, this would require accessing sys.body_mass inside the function rand. Any leads on why this error occurs?

jloganolson, 2025-04-15:
+1 just ran into this issue - any guidance would be appreciated

btaba, 2025-04-15:
@avirupdas55 @jloganolson please provide MREs and we can take a look

jloganolson, 2025-04-30:
Went with a workaround i passing eval_env and (I believe mentioned elsewhere) making a new instance afterwards to actually run the eval