google/brax #572: compatibility problem with brax.v1

feiden, 2025-01-17:
**Problem:**
When training a v1 environment I get an error "TypeError: wrap_for_training() got an unexpected keyword argument 'randomization_fn'"

**Minimal reproducible example:**
```
import functools
from brax.training.agents.ppo import train as ppo, networks as ppo_networks
from brax.v1.envs import _envs

eval_env = _envs["walker2d"]()
train = functools.partial(ppo.train, network_factory=functools.partial(ppo_networks.make_ppo_networks, value_hidden_layer_sizes=(64, 64)))
make_inference_fn, params, _ = train(environment=eval_env, wrap_env=True, num_timesteps=1000, episode_length=10)
```

**Proposed solution:**
The different `train` functions have a if-else logic to accommodate v1 environments:
```
    if isinstance(environment, envs.Env):
      wrap_for_training = envs.training.wrap
    else:
      wrap_for_training = envs_v1.wrappers.wrap_for_training
```
So they will pass the `randomization_fn` keyword to `envs_v1.wrappers.wrap_for_training` which does not recognize it. A quick fix could be to add `randomization_fn=None` to the signature of `envs_v1.wrappers.wrap_for_training`.

btaba, 2025-01-17:
Hi @feiden , we're not supporting brax v1 anymore (in fact, this reminds me that we need delete those folders altogether)

closed 2025-01-17 by btaba

lupusorina, 2025-02-26:
Any update on when the folders will be deleted? The line `wrap_for_training = envs_v1.wrappers.wrap_for_training` should also be deleted because the function arguments don't take `randomization_fn`. Thanks!