google/brax #574: Reward and termination in `inverted_double_pendulum` environment is incorrect

nic-barbara, 2025-01-30:
Hey Brax Team,

I've been playing around with the `inverted_double_pendulum` environment and I think there are a couple of bugs related to the reward calculations and termination condition. 

### Termination condition

The current code used to calculate the $x$ and $y$ position of the tip of the second pole is this:
```python
tip = base.Transform.create(pos=jnp.array([0.0, 0.0, 0.6])).do(
    pipeline_state.x.take(2)
)
x, _, y = tip.pos
```
As far as I can tell, this does not correctly compute the tip of the second pole. For example, here's a video of a (roughly) trained policy with the current environment implementation. As you can see, it actually doesn't learn to balance the second pole because the environment doesn't terminate, so it keeps receiving a `+10` healthy reward bonus during training.

https://github.com/user-attachments/assets/8738b7ba-e510-443c-99e9-6a6df782ad70

In the [`gymnaisum` implementation](https://github.com/Farama-Foundation/Gymnasium/blob/main/gymnasium/envs/mujoco/inverted_double_pendulum_v5.py#L186), they compute the pole tip position as:
```python
x, _, y = self.data.site_xpos[0]
```
which makes sense given there's already a site set up for the pole tip in the `.xml` file. Switching to this implementation produces what I'd expect:

https://github.com/user-attachments/assets/751ec675-b417-4be8-8da4-09a4043a24a4

### Reward

This is a minor one, but comparing to the `gymnasium` implementation again, the `alive_bonus` should only be added if the environment is not terminated. Proposing changing [this line](https://github.com/google/brax/blob/main/brax/envs/inverted_double_pendulum.py#L165) to this:
```python
done = jnp.where(y <= 1, jnp.float32(1), jnp.float32(0))
reward = (1 - done) * self._alive_bonus - dist_penalty - vel_penalty
```

Happy to submit a PR if you like, let me know!

btaba, 2025-01-30:
Hi @nic-barbara , 

Thanks for digging into this! However, only MJX updates `self.data.site_xpos`. You'll have to update the code to get the tip position manually using `pipeline_state.x.take(2)` to support all brax physics backends. It might just need to be:

```
pipeline_state.x.take(2).do(base.Transform.create(pos=jnp.array([0.0, 0.0, 0.6])))
```

The second video terminates right away, is the wrong video uploaded?

nic-barbara, 2025-01-30:
Ok thanks, I'll give it a crack.

Nope, the second video is what I meant to upload. It now terminates after a few steps (as it should) because I fixed the termination condition. It's using exactly the same policy.

closed 2025-01-31 by btaba