google/brax #587: `error: uninstall-distutils-installed-package` in the "Brax Training with MuJoCo XLA - MJX" colab

samuela, 2025-03-24:
The README links to [this colab](https://colab.sandbox.google.com/github/google-deepmind/mujoco/blob/main/mjx/tutorial.ipynb) which produces this error:

```
Collecting mujoco
  Downloading mujoco-3.3.0-cp311-cp311-manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata (44 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 44.4/44.4 kB 1.1 MB/s eta 0:00:00
Requirement already satisfied: absl-py in /usr/local/lib/python3.11/dist-packages (from mujoco) (1.4.0)
Requirement already satisfied: etils[epath] in /usr/local/lib/python3.11/dist-packages (from mujoco) (1.12.2)
Collecting glfw (from mujoco)
  Downloading glfw-2.8.0-py2.py27.py3.py30.py31.py32.py33.py34.py35.py36.py37.py38.p39.p310.p311.p312.p313-none-manylinux_2_28_x86_64.whl.metadata (5.4 kB)
Requirement already satisfied: numpy in /usr/local/lib/python3.11/dist-packages (from mujoco) (2.0.2)
Collecting pyopengl (from mujoco)
  Downloading PyOpenGL-3.1.9-py3-none-any.whl.metadata (3.3 kB)
Requirement already satisfied: fsspec in /usr/local/lib/python3.11/dist-packages (from etils[epath]->mujoco) (2025.3.0)
Requirement already satisfied: importlib_resources in /usr/local/lib/python3.11/dist-packages (from etils[epath]->mujoco) (6.5.2)
Requirement already satisfied: typing_extensions in /usr/local/lib/python3.11/dist-packages (from etils[epath]->mujoco) (4.12.2)
Requirement already satisfied: zipp in /usr/local/lib/python3.11/dist-packages (from etils[epath]->mujoco) (3.21.0)
Downloading mujoco-3.3.0-cp311-cp311-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (6.6 MB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 6.6/6.6 MB 43.4 MB/s eta 0:00:00
Downloading glfw-2.8.0-py2.py27.py3.py30.py31.py32.py33.py34.py35.py36.py37.py38.p39.p310.p311.p312.p313-none-manylinux_2_28_x86_64.whl (243 kB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 243.4/243.4 kB 17.7 MB/s eta 0:00:00
Downloading PyOpenGL-3.1.9-py3-none-any.whl (3.2 MB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 3.2/3.2 MB 86.6 MB/s eta 0:00:00
Installing collected packages: pyopengl, glfw, mujoco
Successfully installed glfw-2.8.0 mujoco-3.3.0 pyopengl-3.1.9
Collecting mujoco_mjx
  Downloading mujoco_mjx-3.3.0-py3-none-any.whl.metadata (3.3 kB)
Requirement already satisfied: absl-py in /usr/local/lib/python3.11/dist-packages (from mujoco_mjx) (1.4.0)
Requirement already satisfied: etils[epath] in /usr/local/lib/python3.11/dist-packages (from mujoco_mjx) (1.12.2)
Requirement already satisfied: jax in /usr/local/lib/python3.11/dist-packages (from mujoco_mjx) (0.5.2)
Requirement already satisfied: jaxlib in /usr/local/lib/python3.11/dist-packages (from mujoco_mjx) (0.5.1)
Requirement already satisfied: mujoco>=3.3.0.dev0 in /usr/local/lib/python3.11/dist-packages (from mujoco_mjx) (3.3.0)
Requirement already satisfied: scipy in /usr/local/lib/python3.11/dist-packages (from mujoco_mjx) (1.14.1)
Collecting trimesh (from mujoco_mjx)
  Downloading trimesh-4.6.5-py3-none-any.whl.metadata (18 kB)
Requirement already satisfied: glfw in /usr/local/lib/python3.11/dist-packages (from mujoco>=3.3.0.dev0->mujoco_mjx) (2.8.0)
Requirement already satisfied: numpy in /usr/local/lib/python3.11/dist-packages (from mujoco>=3.3.0.dev0->mujoco_mjx) (2.0.2)
Requirement already satisfied: pyopengl in /usr/local/lib/python3.11/dist-packages (from mujoco>=3.3.0.dev0->mujoco_mjx) (3.1.9)
Requirement already satisfied: fsspec in /usr/local/lib/python3.11/dist-packages (from etils[epath]->mujoco_mjx) (2025.3.0)
Requirement already satisfied: importlib_resources in /usr/local/lib/python3.11/dist-packages (from etils[epath]->mujoco_mjx) (6.5.2)
Requirement already satisfied: typing_extensions in /usr/local/lib/python3.11/dist-packages (from etils[epath]->mujoco_mjx) (4.12.2)
Requirement already satisfied: zipp in /usr/local/lib/python3.11/dist-packages (from etils[epath]->mujoco_mjx) (3.21.0)
Requirement already satisfied: ml_dtypes>=0.4.0 in /usr/local/lib/python3.11/dist-packages (from jax->mujoco_mjx) (0.5.1)
Requirement already satisfied: opt_einsum in /usr/local/lib/python3.11/dist-packages (from jax->mujoco_mjx) (3.4.0)
Downloading mujoco_mjx-3.3.0-py3-none-any.whl (6.7 MB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 6.7/6.7 MB 53.1 MB/s eta 0:00:00
Downloading trimesh-4.6.5-py3-none-any.whl (708 kB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 708.9/708.9 kB 36.2 MB/s eta 0:00:00
Installing collected packages: trimesh, mujoco_mjx
Successfully installed mujoco_mjx-3.3.0 trimesh-4.6.5
Collecting brax
  Downloading brax-0.12.1-py3-none-any.whl.metadata (7.7 kB)
Requirement already satisfied: absl-py in /usr/local/lib/python3.11/dist-packages (from brax) (1.4.0)
Collecting dm_env (from brax)
  Downloading dm_env-1.6-py3-none-any.whl.metadata (966 bytes)
Requirement already satisfied: etils in /usr/local/lib/python3.11/dist-packages (from brax) (1.12.2)
Collecting flask (from brax)
  Downloading flask-3.1.0-py3-none-any.whl.metadata (2.7 kB)
Collecting flask_cors (from brax)
  Downloading flask_cors-5.0.1-py3-none-any.whl.metadata (961 bytes)
Requirement already satisfied: flax in /usr/local/lib/python3.11/dist-packages (from brax) (0.10.4)
Requirement already satisfied: grpcio in /usr/local/lib/python3.11/dist-packages (from brax) (1.71.0)
Collecting gym (from brax)
  Downloading gym-0.26.2.tar.gz (721 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 721.7/721.7 kB 9.8 MB/s eta 0:00:00
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Requirement already satisfied: jax>=0.4.6 in /usr/local/lib/python3.11/dist-packages (from brax) (0.5.2)
Requirement already satisfied: jaxlib>=0.4.6 in /usr/local/lib/python3.11/dist-packages (from brax) (0.5.1)
[... 64 lines cut ...]
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 76.5/76.5 kB 5.5 MB/s eta 0:00:00
Downloading pytinyrenderer-0.0.14-cp311-cp311-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (1.9 MB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.9/1.9 MB 59.5 MB/s eta 0:00:00
Downloading tensorboardX-2.6.2.2-py2.py3-none-any.whl (101 kB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 101.7/101.7 kB 7.8 MB/s eta 0:00:00
Downloading blinker-1.9.0-py3-none-any.whl (8.5 kB)
Downloading gym_notices-0.0.8-py3-none-any.whl (3.0 kB)
Downloading itsdangerous-2.2.0-py3-none-any.whl (16 kB)
Downloading werkzeug-3.1.3-py3-none-any.whl (224 kB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 224.5/224.5 kB 10.9 MB/s eta 0:00:00
Building wheels for collected packages: gym
  Building wheel for gym (pyproject.toml) ... done
  Created wheel for gym: filename=gym-0.26.2-py3-none-any.whl size=827727 sha256=a769a4fae576fd9da0df166d61178da391e201c5805dd666abbcd0eb69adf3f0
  Stored in directory: /root/.cache/pip/wheels/1c/77/9e/9af5470201a0b0543937933ee99ba884cd237d2faefe8f4d37
Successfully built gym
Installing collected packages: pytinyrenderer, gym_notices, Werkzeug, tensorboardX, ml_collections, itsdangerous, gym, blinker, flask, dm_env, flask_cors, jaxopt, brax
  Attempting uninstall: blinker
    Found existing installation: blinker 1.4
error: uninstall-distutils-installed-package

× Cannot uninstall blinker 1.4
╰─> It is a distutils installed project and thus we cannot accurately determine which files belong to it which would lead to only a partial uninstall.
```

when running the first cell.

samuela, 2025-03-24:
Adding `!pip install --ignore-installed blinker` to the top seems to work

samuela, 2025-03-24:
I'm also getting an error when running the code for training a policy on barkour:

```
WARNING:absl:[process=0][thread=MainThread] Skipped cross-host ArrayMetadata validation because only one process is found: process_index=0.
---------------------------------------------------------------------------
AssertionError                            Traceback (most recent call last)
[<ipython-input-26-df658f442769>](https://localhost:8080/#) in <cell line: 0>()
     34 env = envs.get_environment(env_name)
     35 eval_env = envs.get_environment(env_name)
---> 36 make_inference_fn, params, _= train_fn(environment=env,
     37                                        progress_fn=progress,
     38                                        eval_env=eval_env)

1 frames
[/usr/local/lib/python3.11/dist-packages/brax/training/pmap.py](https://localhost:8080/#) in assert_is_replicated(x, debug)
     68   """
     69   f = functools.partial(is_replicated, axis_name='i')
---> 70   assert jax.pmap(f, axis_name='i')(x)[0], debug

AssertionError: None
```

btaba, 2025-03-31:
Hi @samuela , I ran the first few cells of the notebook using a V100 instance and cannot reproduce the install error, but it looks like you found a workaround given your setup. 

RE the AssertionError; None, typically this occurs when the training script produces a NaN. I just ran the barkour training on V100 and also cannot reproduce the issue. Perhaps try increasing float precision (see https://github.com/google-deepmind/mujoco_playground/pull/87#pullrequestreview-2730368519 as an example)

closed 2025-03-31 by btaba

samuela, 2025-05-05:
Hi @btaba , I am not able to reproduce on CPU and T4 instances but I am able to reproduce on v2-8 TPU instances on colab.sandbox.google.com FWIW. I don't have access to V100 instances. Not sure why exactly, but perhaps there are slight differences in the VM images.