google/brax #589: Large VRAM Consumption Even for Simple Simulation

hwang-warren, 2025-04-05:
Hi, first thank you for bridging MuJoCo to GPU.

I'm using Nvidia RTX 4090 GPU, and I tried to run this simple test case copied from the [tutorial colab](https://colab.research.google.com/github/google-deepmind/mujoco/blob/main/mjx/tutorial.ipynb).

I noticed that even for such simple simulation (just one swinging articulation), it consumes 80% of my VRAM as shown in the `nvidia-smi -l 1` terminal output, in thew middle `Memory-Usage` section.

Is this still the case for the simulation used in `brax`, or did I use the wrong simulation interface or missed some tricks?

Looking forward to your feedback and thank you for your time!

```python
import mujoco
from mujoco import mjx
import mujoco.viewer
import mediapy as media
import jax
from etils import epath

xml = """
<mujoco>
  <worldbody>
    <light name="top" pos="0 0 1"/>
    <body name="box_and_sphere" euler="0 0 -30">
      <joint name="swing" type="hinge" axis="1 -1 0" pos="-.2 -.2 -.2"/>
      <geom name="red_box" type="box" size=".2 .2 .2" rgba="1 0 0 1"/>
      <geom name="green_sphere" pos=".2 .2 .2" size=".1" rgba="0 1 0 1"/>
    </body>
  </worldbody>
</mujoco>
"""

# Make CPU mujoco model & data from xml
mj_model = mujoco.MjModel.from_xml_string(xml)
mj_data = mujoco.MjData(mj_model)

# Init system configuration
# Here we choose to: Reset data
mujoco.mj_resetData(mj_model, mj_data)

# Put model and data on GPU
mjx_model = mjx.put_model(mj_model)
mjx_data = mjx.put_data(mj_model, mj_data)

# Create the renderer
renderer = mujoco.Renderer(mj_model)
# with these options
scene_option = mujoco.MjvOption()
scene_option.flags[mujoco.mjtVisFlag.mjVIS_JOINT] = True  # (optional) Visualize joints
# using this frame rate
framerate = 60  # [fps]

# Setup step function for just-in-time compilation with XLA
jit_step = jax.jit(mjx.step)

# Simulate and render
duration = 3.8  # [seconds]
frames = []
while mjx_data.time < duration:
  # Step the simulation forward
  mjx_data = jit_step(mjx_model, mjx_data)
  
  # Render & store frames according to framerate
  if len(frames) < mjx_data.time * framerate:
    mj_data = mjx.get_data(mj_model, mjx_data)
    renderer.update_scene(mj_data, scene_option=scene_option)
    pixels = renderer.render()
    frames.append(pixels)

# Write video
media.write_video(str(epath.Path(__file__).parent / 'sim_video_rec_by_colab_tutorial.mp4'), frames, fps=framerate)
```

[image]

lvjonok, 2025-04-06:
Hey, @hwang-warren I believe it happens because of jax gpu memory allocation.

Check: https://docs.jax.dev/en/latest/gpu_memory_allocation.html

hwang-warren, 2025-04-06:
Thank you for the help. In my case the following helps reduce preallocation to an acceptable and adjustable level
```python
import os

os.environ["XLA_PYTHON_CLIENT_MEM_FRACTION"] = "0.2"  # preallocated 20% GPU memory
```

lvjonok, 2025-04-06:
> Thank you for the help. In my case the following helps reduce preallocation to an acceptable and adjustable level
> 
> import os
> 
> os.environ["XLA_PYTHON_CLIENT_MEM_FRACTION"] = "0.2"  # preallocated 20% GPU memory

Sometimes it may be useful to disable preallocation at all like this:

```python
os.environ["XLA_PYTHON_CLIENT_PREALLOCATE"] = "false"
```

btaba, 2025-04-07:
Closing the issue since it looks like the main concern has been resolved!

closed 2025-04-07 by btaba