google/brax #618: Incompatibility with mujoco v3.3.3

GaetanLepage, 2025-06-12:
Context: [bumping mujoco in nixpkgs](https://github.com/NixOS/nixpkgs/pull/415961).

With `mujoco==3.3.3`, the following tests fail:
```
=========================== short test summary info ============================
FAILED brax/mjx/perf_test.py::PerfTest::test_pipeline_ant - TypeError: Data.init() got an unexpected keyword argument 'contact'
FAILED brax/mjx/pipeline_test.py::PipelineTest::test_pendulum - TypeError: Data.init() got an unexpected keyword argument 'contact'
FAILED brax/mjx/pipeline_test.py::PipelineTest::test_pipeline_init_with_ctrl - TypeError: Data.init() got an unexpected keyword argument 'contact'
=========== 3 failed, 894 passed, 411 warnings in 860.93s (0:14:20) ============
```

This can be related to [this change](https://github.com/google-deepmind/mujoco/commit/6bc0bbf7df3058ebfe384022c880ff96239ea1d3) made upstream. (There is of course more to it than this single commit).

erikfrey, 2025-06-13:
Hi @GaetanLepage - we just released [Brax v0.12.4](https://pypi.org/project/brax/0.12.4/) which should address your problem.  Please reopen the issue if you're still seeing any issues.

GaetanLepage, 2025-06-13:
Indeed, it works great: https://github.com/NixOS/nixpkgs/pull/416535.
Thanks a lot!