google/brax #633: Collision failures frequent among primitives: box-box, box-capsule, capsule-capsule, etc

fionalluo, 2025-08-25:
### My setup

Brax: Cloned from https://github.com/google/brax/commits/main/, from the August 21st commit
CUDA: 12.6
GPU: NVIDIA GeForce RTX 3090
MuJoCo version: 2.3.7
Architecture: x86_64

### What's happening? What did you expect?

I have been trying to construct several environment using the Leap Hand model, which I got from [MuJoCo playground](https://github.com/google-deepmind/mujoco_playground). MuJoCo playground only provides tasks where the Leap Hand manipulate a cube (like spin z-axis or reorient), the collisions work fine for that task. 

However, when I tried to construct new Leap Hand environments including opening a door and pick-place of a cube, I found the **collisions**  to be very faulty, with the leap hand able to pass through the door, cube, and most objects. I found **box-box**, **box-capsule**, **capsule-capsule**, **sphere-box**, etc collisions to all be faulty with objects passing through each other. Note that in this new task, the Leap Hand has additional sliding joints allowing it to move in 3D.

I have already tried several fixes, which have been unsuccessful, including

- Replacing all the box collisions on the hand to capsules of the same size. I did this because I wondered if box collisions were uniquely faulty, however capsules had issues too (though they seemed to have less collision failures)
- Reducing the velocity of the hand when colliding with object, in case the velocity was too high
- Replacing the sliding joints with a robot arm, in case the sliding joints were the problem
- Increasing `num_substeps`, from the default in MuJoCo playground of 5 to 50 and 500. This is equivalent to reducing `sim_dt` from 0.01 to 0.001 and 0.0001.
- Clipping actions to stay within the `ctrl_range`
- Trying different jax versions, including 0.7.0 and 0.5.0
- Rewriting the xml files to have the exact same structure/properties as those in the MuJoCo playground cube task files.
- Adding spheres to the fingertips of each finger on the hand, however this can still pass through the door

https://github.com/user-attachments/assets/c96b3bab-1434-4e3a-951c-eacb52224932

https://github.com/user-attachments/assets/2677faab-50c3-412c-8404-33b37e258fa1

https://github.com/user-attachments/assets/2524a817-bcc9-4751-8b43-f19a1322b449

### Steps for reproduction

1. Clone my copy of [MuJoCo playground](https://github.com/edwhu/mujoco_playground)
2. Follow the UV environment setup in the README. It's the same as the mujoco playground setup.
3. Switch to the branch fiona/issue, `git checkout fiona/issue`
4. Run `python render_test_edward.py`. This script will give random actions to collide the hand against the door. It will take a minute to run.
Watch the saved video, `test_env_collision.mp4`. It will show the hand passing through the door repeatedly.

### Code required for reproduction

The few relevant files from steps for reproduction are linked below, since the repo is large. 
1. [Leap Hand XML](https://github.com/edwhu/mujoco_playground/blob/fiona/issue/mujoco_playground/_src/manipulation/leap_hand/xmls/door_leap_box.xml)
2. [Door XML](https://github.com/edwhu/mujoco_playground/blob/fiona/issue/mujoco_playground/_src/manipulation/leap_hand/xmls/door.xml)
3. [Scene XML](https://github.com/edwhu/mujoco_playground/blob/fiona/issue/mujoco_playground/_src/manipulation/leap_hand/xmls/door_scene.xml)
4. [Environment Python Class](https://github.com/edwhu/mujoco_playground/blob/fiona/issue/mujoco_playground/_src/manipulation/leap_hand/door_open.py)
5. [Test Render Script](https://github.com/edwhu/mujoco_playground/blob/fiona/issue/render_test_edward.py)

### Confirmations

- [x] I searched the latest documentatino thoroughly before posting.
- [x] I searched previous issues, I am certain this has not been raised before.

### Packages
Here's a more detailed list of my packages
Here's a more detailed list of every package in my uv environment.

Package                   Version        Editable project location
------------------------- -------------- -----------------------------------
absl-py                   2.3.1
aiofiles                  24.1.0
annotated-types           0.7.0
anyio                     4.10.0
argon2-cffi               25.1.0
argon2-cffi-bindings      25.1.0
arrow                     1.3.0
astroid                   3.3.11
asttokens                 3.0.0
async-lru                 2.0.5
attrs                     25.3.0
[... 178 lines cut ...]
terminado                 0.18.1
tinycss2                  1.4.0
toml                      0.10.2
tomlkit                   0.13.3
toolz                     1.0.0
tornado                   6.5.2
tqdm                      4.67.1
traitlets                 5.14.3
treescope                 0.1.10
trimesh                   4.7.4
types-python-dateutil     2.9.0.20250809
typing-extensions         4.14.1
typing-inspection         0.4.1
uri-template              1.3.0
urllib3                   2.5.0
virtualenv                20.34.0
wandb                     0.21.1
wcwidth                   0.2.13
webcolors                 24.11.1
webencodings              0.5.1
websocket-client          1.8.0
werkzeug                  3.1.3
widgetsnbextension        4.0.14
wrapt                     1.17.3
zipp                      3.23.0

btaba, 2025-08-25:
Hi @fionalluo , have you fiddled with these settings

https://github.com/edwhu/mujoco_playground/blob/fe23c345ebc889f1141873424793e2116bfdeea2/mujoco_playground/_src/manipulation/leap_hand/xmls/door_leap_box.xml#L8-L11

or contype/conaffinity?

You can also try the warp implementation if you don't want to fiddle with the sharp bits of the JAX implementation. See the discussion [here](https://github.com/google-deepmind/mujoco_warp/issues/639)

btaba, 2025-08-25:
Closing this as duplicate...

closed 2025-08-25 by btaba

fionalluo, 2025-08-25:
Thank you so much for the prompt response. I will reply on the open thread: [https://github.com/google-deepmind/mujoco_playground/issues/191](https://github.com/google-deepmind/mujoco_playground/issues/191)